sajmons/CollimationCircles #1: cannot run

Agwising, 2023-01-03:
hi! i have tried installing dotnet however i don't know if i did it right. here's my latest error:

~/CollimationCircles/CollimationCircles$ dotnet run
/usr/lib/dotnet/sdk/6.0.111/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(144,5): error NETSDK1045: The current .NET SDK does not support targeting .NET 7.0.  Either target .NET 6.0 or lower, or use a version of the .NET SDK that supports .NET 7.0. [/home/agwising/CollimationCircles/CollimationCircles/CollimationCircles.csproj]

The build failed. Fix the build errors and run again.

then i tried to build it then i got this error:

~/CollimationCircles/CollimationCircles$ dotnet publish --runtime linux-arm64 /p:PublishSingleFile=true --output ~/Home/CollimationCircles
Microsoft (R) Build Engine version 17.0.1+b177f8fa7 for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
/usr/lib/dotnet/sdk/6.0.111/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(144,5): error NETSDK1045: The current .NET SDK does not support targeting .NET 7.0.  Either target .NET 6.0 or lower, or use a version of the .NET SDK that supports .NET 7.0. [/home/a******/CollimationCircles/CollimationCircles/CollimationCircles.csproj]

please help thank you!

Agwising, 2023-01-03:
already found the fix! replaced the version 7.0 to 6.0 on the csproj file. thanks!

sajmons, 2023-01-03:
I'm glad that you managed to find solution for your problem.
There are two ways to fix your problem. either install .NET 7 or change .NET7 to .NET6 in csproj file as you already figure out.

closed 2023-01-03 by sajmons