sajmons/CollimationCircles #2: Correction of camera chip offset?

energiebrand, 2023-01-26:
Hi, 
thanks for the next version of the collimator! I have published the new version in a German astro-forum.
There is a user question about the comparison to the OCAL commercial collimator product.
The OCAL gives the possibility to adjust a camera`s offset if the chip is not installed in the center of it`s housing.

Does CollimationCircles have or will have such a feature? Don´t ask me how it is done, it`s just a question... ;-)

Cheers
Hartmut

sajmons, 2023-01-26:
Can you provide some documentation source describing that functionality so that I can check it?

sajmons, 2023-01-27:
It's my humble opinion that center of camera sensor does not affect telescope optical collimation in any way. Maybe I'm totaly wrong but it doesn't feels logical. My opinion is, that if camera sensor is off a little, this only means that final image will be offset a little too, but it should not have any effect to image quality. Please correct me if I'm wrong.

energiebrand, 2023-01-28:
Hi, thanks for your feedback. I think you are right! Maybe it is only a marketing argument, a feature which should tell the customer that their product is perfect for collimation. But to be frank, it will only reach the last 0,001%. However that would require that my optical adjustment with my eyes is 100% perfect!
So I think I will close this issue,... never was one, just a clarification :-)

closed 2023-01-28 by energiebrand