sajmons/CollimationCircles #3: window movement (and transparency)

DerPit, 2023-01-28:
More of a general feedback.
Just downloaded the latest binary release (1.0.3) and installed the dotnet-runtime on my system (openSUSE Tumbleweed; MS only has a repo for Leap, but it works fine).

I use KDE as DE, running on top of X11.  There, transparency only works if compositing is enabled (can be toggled with Alt-Shift-F12)

What does not work for me is the positioning of the window with cursor keys.  Instead of moving one pixel in the direction intended, it only jumps a full height of the window top bar down (regardless which direction key I pressed).  It is probably is coming from KDE/plasma itself, I've seen that with other applications like the XV image viewer.

Apart from that it seems like a really flexible and easy tool with a clean interface.  Great work.
Can't test it for real collimation, as I'm away from my telescopes until April...

sajmons, 2023-01-28:
Thank you very much foe your feedback. It's great to hear that folks is using application and that it's working quite well on so many platforms.

I have noticed wierd window movement issue too. I openec issue on avaloniaUi and they marked it as a bug. You can see it here: https://github.com/AvaloniaUI/Avalonia/issues/9972

sajmons, 2023-06-23:
Window movement issue was fixed by Avalonia team in 11 preview version. I'm closing this issue.

closed 2023-06-23 by sajmons