sajmons/CollimationCircles #4: Can’t work on MacOS

teoteo, 2023-06-16:
Hi, I’m trying to run on a fresh virtual machine with MacOS 14.4 on Apple Silicon (arm) processor.
I have “Unhandled exception. System. NullReferenceException: Object reference not set to an instance of an object.” errors.
[image]

There is a little typo in readme file to:
`cd ColiminationCircles/ColiminationCircles`
should be 
`cd ColliminationCircles/ColliminationCircles`

sajmons, 2023-06-16:
@teoteo Thank you trying to run it on macOS. I will investigate that and hopefully I can make it to work on macOS too.

sajmons, 2023-06-29:
@teoteo In latest release 2.1.0 there is build fro macosx 64bit. Maybe you can try it and tell me if it works?

mikegeyser, 2023-06-29:
I'm not sure if it's relevant, but I found [this](https://docs.avaloniaui.net/docs/distribution-publishing/macos) on the Avalonia website regarding Mac redistributable apps. The bundle doesn't seem to match this structure? 

(Please ignore if this isn't relevant.)

sajmons, 2023-06-30:
@mikegeyser Thank you for your report. It turns out that there is bug in latest release that causes window not transparent. I'll try to fix that as soon as posible. If you know how, you can try to build version 2.0.1 to see if window is transparent there.

I'll investigate link that you send it's full of useful information. In future releases I'll try to pack mac version in th right way. maybe you can try to download only ZIP, change permission on CollimationCircles file to make it executable with command chmod +x ./CollimationCircles. I think it should work on mac too. You will need admin permissions to run that command.

mikegeyser, 2023-06-30:
Cool, will try both of those things and let you know!

mikegeyser, 2023-06-30:
Building and running version 2.0.1, the background still seems to be opaque. 

[image]

When I chmod the file and run it, it starts executing - but then prompts with a security warning. I'll do some googling and see if it's something I can bypass on my side. 

[image]

Here is the full exception written to stderr:

```
Unhandled exception. System.TypeInitializationException: The type initializer for 'SkiaSharp.SKImageInfo' threw an exception.
 ---> System.DllNotFoundException: Unable to load shared library 'libSkiaSharp' or one of its dependencies. In order to help diagnose loading problems, consider setting the DYLD_PRINT_LIBRARIES environment variable:
dlopen(/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp.dylib, 0x0001): tried: '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp.dylib' (code signature in <19A649A7-94C0-3B2D-8797-C327D9F3B9C8> '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp.dylib' not valid for use in process: library load disallowed by system policy), '/System/Volumes/Preboot/Cryptexes/OS/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp.dylib' (no such file), '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp.dylib' (code signature in <19A649A7-94C0-3B2D-8797-C327D9F3B9C8> '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp.dylib' not valid for use in process: library load disallowed by system policy)
dlopen(libSkiaSharp.dylib, 0x0001): tried: 'libSkiaSharp.dylib' (code signature in <19A649A7-94C0-3B2D-8797-C327D9F3B9C8> '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp.dylib' not valid for use in process: library load disallowed by system policy), '/System/Volumes/Preboot/Cryptexes/OSlibSkiaSharp.dylib' (no such file), '/usr/lib/libSkiaSharp.dylib' (no such file, not in dyld cache), 'libSkiaSharp.dylib' (code signature in <19A649A7-94C0-3B2D-8797-C327D9F3B9C8> '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp.dylib' not valid for use in process: library load disallowed by system policy), '/usr/local/lib/libSkiaSharp.dylib' (no such file), '/usr/lib/libSkiaSharp.dylib' (no such file, not in dyld cache)
dlopen(/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/liblibSkiaSharp.dylib, 0x0001): tried: '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/liblibSkiaSharp.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/liblibSkiaSharp.dylib' (no such file), '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/liblibSkiaSharp.dylib' (no such file)
dlopen(liblibSkiaSharp.dylib, 0x0001): tried: 'liblibSkiaSharp.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSliblibSkiaSharp.dylib' (no such file), '/usr/lib/liblibSkiaSharp.dylib' (no such file, not in dyld cache), 'liblibSkiaSharp.dylib' (no such file), '/usr/local/lib/liblibSkiaSharp.dylib' (no such file), '/usr/lib/liblibSkiaSharp.dylib' (no such file, not in dyld cache)
dlopen(/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp, 0x0001): tried: '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp' (no such file), '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/libSkiaSharp' (no such file)
dlopen(libSkiaSharp, 0x0001): tried: 'libSkiaSharp' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibSkiaSharp' (no such file), '/usr/lib/libSkiaSharp' (no such file, not in dyld cache), 'libSkiaSharp' (no such file), '/usr/local/lib/libSkiaSharp' (no such file), '/usr/lib/libSkiaSharp' (no such file, not in dyld cache)
dlopen(/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/liblibSkiaSharp, 0x0001): tried: '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/liblibSkiaSharp' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/liblibSkiaSharp' (no such file), '/Users/mikegeyser/Downloads/4-CollimationCircles-2.1.0-osx-x64/liblibSkiaSharp' (no such file)
dlopen(liblibSkiaSharp, 0x0001): tried: 'liblibSkiaSharp' (no such file), '/System/Volumes/Preboot/Cryptexes/OSliblibSkiaSharp' (no such file), '/usr/lib/liblibSkiaSharp' (no such file, not in dyld cache), 'liblibSkiaSharp' (no such file), '/usr/local/lib/liblibSkiaSharp' (no such file), '/usr/lib/liblibSkiaSharp' (no such file, not in dyld cache)

   at SkiaSharp.SKImageInfo..cctor()
   --- End of inner exception stack trace ---
   at Avalonia.Skia.SkiaPlatform.Initialize(SkiaOptions options) in /_/src/Skia/Avalonia.Skia/SkiaPlatform.cs:line 20
   at Avalonia.AppBuilder.SetupUnsafe() in /_/src/Avalonia.Controls/AppBuilder.cs:line 302
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 219
   at CollimationCircles.Program.Main(String[] args)
[1]    80594 abort      ./CollimationCircles
```

sajmons, 2023-06-30:
What is your macosx processor arhitecture?

mikegeyser, 2023-06-30:
@sajmons It's 64-bit. I have an Intel Core i7 processor, not the Apple Silicon. 

From my brief googling it looked like this was related to the signing of the dependencies, that 'app' packaging if you will, although I admit I'm a little out of my depth.  😅

sajmons, 2023-06-30:
Yeah, I asume same as you. I have made experimental app boundle, but it's not signed yet. So I asume there will be same issues.
You can download it here: https://drive.google.com/file/d/1PAPp29zrhV9ggrpPn_g_ZP4twDdztYPD/view?usp=sharing

mikegeyser, 2023-06-30:
I tried opening the app bundle, both using Finder (double clicking) and via console. It really doesn't seem happy with it, unfortunately.

[image]

```
The application cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x600001070120 {Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}
```

sajmons, 2023-07-03:
I will try to install masos inside VirtualBox to get that working.

teoteo, 2023-07-03:
I get the same error with the app bundle.
Trying to get it with git clone, I get those errors:
```
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at Avalonia.Native.WindowImpl..ctor(IAvaloniaNativeFactory factory, AvaloniaNativePlatformOptions opts, AvaloniaNativeGlPlatformGraphics glFeature) in /_/src/Avalonia.Native/WindowImpl.cs:line 35
   at Avalonia.Native.AvaloniaNativePlatform.CreateWindow() in /_/src/Avalonia.Native/AvaloniaNativePlatform.cs:line 149
   at Avalonia.Controls.Platform.PlatformManager.CreateWindow() in /_/src/Avalonia.Controls/Platform/PlatformManager.cs:line 31
   at Avalonia.Controls.Window..ctor() in /_/src/Avalonia.Controls/Window.cs:line 214
   at CollimationCircles.Views.MainView..ctor() in /Users/matteoadmin/CollimationCircles/CollimationCircles/Views/MainView.axaml.cs:line 18
   at CollimationCircles.App.OnFrameworkInitializationCompleted() in /Users/matteoadmin/CollimationCircles/CollimationCircles/App.axaml.cs:line 29
   at Avalonia.AppBuilder.SetupUnsafe() in /_/src/Avalonia.Controls/AppBuilder.cs:line 311
   at Avalonia.AppBuilder.Setup() in /_/src/Avalonia.Controls/AppBuilder.cs:line 291
   at Avalonia.AppBuilder.SetupWithLifetime(IApplicationLifetime lifetime) in /_/src/Avalonia.Controls/AppBuilder.cs:line 199
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 219
   at CollimationCircles.Program.Main(String[] args) in /Users/matteoadmin/CollimationCircles/CollimationCircles/Program.cs:line 16
```

sajmons, 2023-07-04:
I have managed to set up macOS as VMVare virtual machine foloving instructions from here https://lauri-elias.medium.com/macos-monterey-on-ryzen-windows-vmware-16-2ebe377d90b3

I hope now I will be able to produce valid macOS package for CollimationCircles app.

sajmons, 2023-07-04:
Hi @mikegeyser and @teoteo.
Here is properly packed app bundle for macOS. https://drive.google.com/file/d/1VZo80ac24YYAxMpNn0qYGzD_g8_bNwwV/view?usp=drive_link

In my virtual machine window is not transparent, maybe on real hardvare it is? Can you try it?

mikegeyser, 2023-07-06:
Ok! I've managed to open the app, but it required some manual steps. When I first tried to open it, I received this error: `The application cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed."`

So I ran the following chmod command into the app bundle: `chmod +x CollimationCircles\ 2.app/Contents/MacOS/CollimationCircles` to give the actual executable run priveleges.

Then it prompted with the typical MacOS 'This publisher can't be trusted' security warning, which I closed and went to Settings > Privacy and clicked to allow the app to run. Then when I opened it again it let me click 'Open Anyway' when the security warning came up. (This last step is pretty typical of unsigned apps on MacOS)

The window, unfortunately, isn't transparent. I looked at the Avalonia repo, and it seems like there may have been a problem with [11 preview 5 and transparency on MacOS](https://github.com/AvaloniaUI/Avalonia/pull/10196)? But that was in February, and I see that the app is on Avalonia 11.0.0-rc2.2, which should surely have those fixes in it.

[image]

mikegeyser, 2023-07-06:
Aha! I took a guess based off of that PR, and added `TransparencyLevelHint="Transparent"` onto the `Window` object in `MainView.axaml` and it seems to make the main window transparent!

[image]

sajmons, 2023-07-06:
@mikegeyser many thanx for pointing this out. I have uploaded new app bundle to same location https://drive.google.com/file/d/1VZo80ac24YYAxMpNn0qYGzD_g8_bNwwV/view?usp=drive_link.

Please give it a try again. And let me know if transparency now works out of the box?

sajmons, 2023-07-06:
@teoteo latest version located here https://drive.google.com/file/d/1VZo80ac24YYAxMpNn0qYGzD_g8_bNwwV/view?usp=drive_link should work now. By double clicking application. Window should be transparent now too.

Your feedback would be greatly appriciated.

mikegeyser, 2023-07-06:
@sajmons I still had to run the chmod command before it would execute, but the transparency works!

[image]

sajmons, 2023-07-06:
@mikegeyser Yeah, I know, chmod is necessary for now. As far as I know, to avoid that, application would need to be build on linux or unix like system. Then add required permissions and than package it using tar. I'll try to do that with WSL (Windows Subsystem for Linux) or maybe transfer app to macOS, set permisions and repack it there. It's a lot of work to be done on every release though. I try to automate it somehow, maybe using GitHub actions (but those don't support arm64 needed for Raspberry PI build). :(

But for now I'm verry happy that with your help I managed to get it working on macOS too.

sajmons, 2023-07-07:
This could be useful too: https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/mac

I'll try to experiment with this too:

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>com.apple.security.cs.allow-jit</key>
    <true/>
    <key>com.apple.security.cs.allow-unsigned-executable-memory</key>
    <true/>
    <key>com.apple.security.cs.disable-library-validation</key>
    <true/>
    <key>com.apple.security.cs.allow-dyld-environment-variables</key>
    <true/>
    <key>com.apple.security.app-sandbox</key>
    <true/>
    <key>com.apple.security.temporary-exception.mach-lookup.global-name</key>
    <array>
        <string>com.apple.coreservices.launchservicesd</string>
    </array>
</dict>
</plist>

mikegeyser, 2023-07-20:
Hey @sajmons, I just wanted to let you know that I managed to successfully use CollimationCircles on my Mac to collimate my telescope last night. I had to design and print a ghetto bracket for my Logitech StreamCam, but it made the process much simpler overall, and I think it's in far better collimation now. I just need to figure out how to disable the autofocus on the webcam to get it more precise.

Thank you very much for all of the effort on this application, it's so appreciated.

[image]

sajmons, 2023-07-21:
Wow! It looks like your collimation is pin point accurate!
You can go one step further from here. Point your telescope to a bright
star, put a bahtinov mask on and then use the bahtinov mask pattern in the
Collimation Circles app to see if there are any imperfections left.

Great yob!

Can you please send me a screenshot of your macOS desktop with Collimation
Circles running? With your permission I would like to use that image on my
website.
Can I mention your name too?

Best regards,
Simon

sajmons, 2023-07-21:
i think this issue is resolved now. Application is fully working on macOS too.

closed 2023-07-21 by sajmons

mikegeyser, 2023-07-21:
Would this suffice as a screenshot @sajmons? You're more than welcome to use the image (and mention my name, but you really don't need to).

[image]
[image]

sajmons, 2023-07-21:
Thank you Mike! Your images are great!
See blog post on my website:
https://saimons-astronomy.webador.com/1403400_collimation-circles-macos-breaktrough

Best regards,
Simon