sajmons/CollimationCircles #6: packaging for OS

MattBlack85, 2023-06-21:
Hi @sajmons 

I had a go with the app and I can compile on Linux and it works (still to try on arm but I do not expect any issue)

What I would like to achieve tho is a ready to be installed app for ArchLInux (x64 and aarch64) and here I would need some guidance because I am a .NET dumb (never done in my life).

I would need a step by step guide to reach a point where a given folder containing the needed bits can be unpacked on the disk to the right place in order to work.

Specifically:

- I see there are two libraries that are shipped with the zip packages, are you bundling them by yourself when packaging? Could you add them (in all fashions) to the repo so it will be easier to bundle them?
- I see an appstate..json is also bundled within the zip archive, where that file should live given, that the executable will be under `/usr/bin`?
- the archive contains also a `CollimationCircles.pdb`, same question as above ^

Thanks!

Mattia

sajmons, 2023-06-21:
Hi @MattBlack85 

I'm using these commands for my builds:

**win-x64**
```
dotnet restore .\CollimationCircles.sln -r win-x64
dotnet publish -c Release -f net6.0 -r win-x64 -o D:\Projekti\Publish\cc\win-64 --self-contained true /p:PublishSingleFile=true /p:PublishReadyToRun=true
```

**linux-x64**
```
dotnet restore .\CollimationCircles.sln -r linux-x64
dotnet publish -c Release -f net6.0 -r linux-x64 -o D:\Projekti\Publish\cc\linux-64 --self-contained true /p:PublishSingleFile=true /p:PublishReadyToRun=true
```

**linux-arm64**
```
dotnet restore .\CollimationCircles.sln -r linux-arm64
dotnet publish -c Release -f net6.0 -r linux-arm64 -o D:\Projekti\Publish\cc\linux-arm64 --self-contained true /p:PublishSingleFile=true /p:PublishReadyToRun=true
```

Commands from above will do all the necessary stuf for you and you get packed app in -o directory. You must have .NET SDK installed!

Appstate.json is not important for distribution and it's created by the app on exit to store current state. It is loaded again when app starts. App should have write permission to the folder where executable is located in order to create appstate.json.

CollimationCircles.pdb contains some debugging symbols for .NET and it is not important, probably there is some flag when publish to remove it. It can be deleted and app should start anyway.

I hope I provided you with all the answers. If not, don't hasistate to ask.

MattBlack85, 2023-06-22:
all cool, thanks @sajmons I think I should be able to make this available by today as a Arch package!

closed 2023-06-22 by MattBlack85