sajmons/CollimationCircles #7: [LINUX] strip breaks the binary

MattBlack85, 2023-06-22:
Hi @sajmons 

I found a weird issue which I am not sure where it originates from since I am a .NET sheep.

So, everything works correctly if I run the commands you gave me and the app starts, but when packaging for Arch the package got broken.

After some investigations I found out that stripping symbols completely breaks the executable.

[`strip`](https://www.man7.org/linux/man-pages/man1/strip.1.html) is generally used to remove unneeded symbols from libs/bins and it helps making also the program smaller (122MB not stripped VS 21MB stripped)

do you know if there is a way to strip safely an executable generated with the .NET sdk?

in the meanwhile I will skip stripping for the ArchLinux package

sajmons, 2023-06-23:
I guess that .NET eqvivalent of strip would be trim.

You can read about it here: https://ivanderevianko.com/2020/09/make-apps-small-again-trimming-net-5

But as my knowledge currently only trimmode CopyUsed working with AvaloniaUI.
See comment in CollimationCircles.csproj file: 

More here:

- https://github.com/AvaloniaUI/Avalonia/issues/9127
- https://github.com/AvaloniaUI/Avalonia/issues/6892

You can try with trimmode CopyUsed but it's not guaranted to work either. You can further limit trimming with adding tags in csproj file as described here: https://ivanderevianko.com/2020/09/make-apps-small-again-trimming-net-5.

In future versions I will try to publish trimmed application. Will addit to TODO list.

Hope that that helps.

sajmons, 2023-06-29:
Good news Mattia! I have revisited my publish workflow and discovered nice option for C# project file <EnableCompressionInSingleFile>true</EnableCompressionInSingleFile>. Enabling this option drastically reduces final executable file size. From 113MB to 51MB. I think this is all I can do. I think Linux strip command can not be used with dotnet apps.
See this in action in next version.

closed 2023-06-29 by sajmons

sajmons, 2023-06-29:
Im using this command now for publishing:

dotnet publish -c Release -f net7.0 -r linux-arm64 -o d:/Projekti/Publish/linux-arm64 --self-contained true /p:PublishSingleFile=true /p:PublishReadyToRun=true /p:DebugType=None /p:DebugSymbols=false

MattBlack85, 2023-06-29:
@sajmons thanks! I will update my build script ASAP :rocket:

sajmons, 2023-06-29:
@MattBlack85 I have just released new version 2.1.0

MattBlack85, 2023-06-29:
yup, already compiling and packaging for archlinux :)

sajmons, 2023-06-29:
Great! Thank you very much!