sajmons/CollimationCircles #8: version 2.1.0 lost transparency on KDE

MattBlack85, 2023-06-29:
hi @sajmons I compiled the 2.1.0 but I noticed that it lost its transparency when running the app.

I have the compositor enabled and it indeed stays on the top of other windows always, but the BG is black.

I thought it was the new flags for removing the debug symbols, but I compiled 2 versions (with and without) and the effect is the same

sajmons, 2023-06-30:
@MattBlack85 Yeah, you are right. I'll investigate and create new release.
Till then I have removed release 2.1.0 from releases.

sajmons, 2023-06-30:
I commited fix for transparency issue with 2.1.0 version. Please pull changes from GitHub and rebuild it. And please let me know if window is transparent now?

sajmons, 2023-07-03:
@MattBlack85 Can you confirm, that transparency works with version 2.1.1?

MattBlack85, 2023-07-03:
Hi @sajmons, unfortunately I left for holidays on Friday and I won't be back at my compilation hardware till the end of july 🤓

will confirm once I'll be back but I trust your fix 💪

sajmons, 2023-07-11:
@nigeldun thank you for testing it and confirm it works. I will close this issue now.

closed 2023-07-11 by sajmons