sajmons/CollimationCircles #11: utilizes an UVC USB camera

tlkw79, 2023-08-07:
Hi,

Thanks for the great app here. Is it possible to include a UVC USB camera in your app?  For an example, [https://github.com/edu27/pyReflectorCollimator](url)

Regards,
Tom

sajmons, 2023-08-17:
In feature release I'm planing to support displaying camera sttream in window background. But user will still need to provide stream by himself.

tlkw79, 2023-08-17:
That's great! Can't wait for it.

tlkw79, 2023-08-17:
By the way, are you planning to release an Android version or ipad version?

sajmons, 2023-08-18:
Android and iPad versions are not planed for now. Maybe in the future if enough users will show interest in it. AvaloniaUI technology has support for both Android and iOS.

tlkw79, 2023-11-02:
> Android and iPad versions are not planed for now. Maybe in the future if enough users will show interest in it. AvaloniaUI technology has support for both Android and iOS.

There is an android app called OpenLiveStacker is welcoming other developer to collaborate with it.  

[https://github.com/artyom-beilis/OpenLiveStacker](url)

Tom

sajmons, 2024-02-05:
UVC and Rasperry Pi camera stream now available. Closing

closed 2024-02-05 by sajmons