sajmons/CollimationCircles #18: Adding some camera controls, focus, zoom, etc.

sajmons, 2024-02-05:
See here for some documentation
https://www.kurokesu.com/main/2016/01/16/manual-usb-camera-settings-in-linux/
https://code.videolan.org/videolan/vlc/-/issues/11105

tlkw79, 2024-02-23:
Any progress on this? Thanks.

sajmons, 2024-02-23:
Yes Tommy, actually I have beta version available to test.
Camera controls are curently working for UVC cameras on Windows and on Linux. 
Raspberry PI camera controls are not supported yet.
Would you be interested to test it?

tlkw79, 2024-02-23:
Oh yes. I can test the beta version for you since I have both Linux and Windows OS with UVC camera. 

Thanks.

sajmons, 2024-02-23:
Here is beta release that you can test.
https://github.com/sajmons/CollimationCircles/releases/tag/version-3.2.0-beta

Please write all your findings here.
Thank you!

tlkw79, 2024-02-23:
Hi Sajmons,

This is what I found out in the beta release.

Kubuntu:
1. Brightness until Zoom functions are okay to use.
2. Sadly Gain until Exposure functions are nothing happen when adjusting. Especially the FOCUS, nothing happens at all.
3. The camera control windows are always kept below others. A bit inconvenient. 
4. when the UVC camera started, the framerates became very slow. 
5. I have to always adjust the Camera Video stream windows to the correct size every time CC starts. 

Windows OS:
1. The camera control windows are always kept below others. 
2. Camera control nothing working in Windows. 

these are what I've found so far.  

Thanks.

sajmons, 2024-02-23:
What version of windows were you using when testing?

tlkw79, 2024-02-23:
Windows 10

sajmons, 2024-02-23:
Interesting on windows 11 controls are working. I'll investigate.

tlkw79, 2024-02-23:
Ok. Thanks a lot.  

Maybe you can consider creating a discord group for CC discussion?

tlkw79, 2024-02-23:
One more issue on my Kubuntu Laptop with webcam. Even though UVC has already been selected, when I start the live stream, it always connects to my laptop webcam.

tlkw79, 2024-02-23:
Okay, I tested my UVC  camera in Windows 11. Nothing happens with camera control too. Anything that I need to add or install?

sajmons, 2024-02-23:
> One more issue on my Kubuntu Laptop with webcam. Even though UVC has already been selected, when I start the live stream, it always connects to my laptop webcam.

Probably your camera is UVC camera is not /dev/video0 device. Use `v4l2-ctl --list-devices` command in terminal to see your UVC camera.

Then try to change v4l2:///dev/video0 address to v4l2:///dev/video1 or what is aplicable for your UVC camera. I should add here dropdown of available cameras.

tlkw79, 2024-02-24:
> > Okay, I tested my UVC camera in Windows 11. Nothing happens with camera control too. Anything that I need to add or install?
> 
> Hmm, UVC cameras should work out of the box. Do you have multiple cameras on your Windows system?

there is a webcam in my notebook.

tlkw79, 2024-02-25:
Now Kubuntu and Windows 10/11 manage to display via my UVC camera. But both still cannot control the camera parameters. Going through other webcam software no problem. 

Under Kubuntu, my webcam display becomes very lag.

sajmons, 2024-03-05:
Can you please confirm if command line controls are working, while CC is running on linux?

Example:
v4l2-ctl \
--set-ctrl=brightness=150 \
--set-ctrl=contrast=51 \
--set-ctrl=saturation=32 \
--set-ctrl=white_balance_temperature_auto=0 \
--set-ctrl=gain=90 \
--set-ctrl=power_line_frequency=1 \
--set-ctrl=white_balance_temperature=1140 \
--set-ctrl=sharpness=24 \
--set-ctrl=backlight_compensation=1 \
--set-ctrl=exposure_auto=1 \
--set-ctrl=exposure_absolute=870 \
--set-ctrl=exposure_auto_priority=1

tlkw79, 2024-03-06:
> Can you please confirm if command line controls are working, while CC is running on linux?
>
> Example: v4l2-ctl --set-ctrl=brightness=150 --set-ctrl=contrast=51 --set-ctrl=saturation=32 [...]
 
Hi,

This is the result i got from console.

ommylkw@tommylkw-N3-Linux:~$ v4l2-ctl --set-ctrl=brightness=150
unknown control 'brightness'
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl --set-ctrl=contrast=52 
unknown control 'contrast'

tlkw79, 2024-03-06:
> exposure_auto_priority=1

From my console, this is what i get. Those commands without unknown control return are good. 

tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl brightness=150
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl contrast=51
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl saturation=32
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl white_balance_temperature_auto=0
unknown control 'white_balance_temperature_auto'
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl gain=90           
unknown control 'gain'
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl power_line_frequency=1
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl white_balance_temperature=1140
VIDIOC_S_CTRL: failed: Permission denied
white_balance_temperature: Permission denied
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl sharpness=24      
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl backlight_compensation=1
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl exposure_auto=1   
unknown control 'exposure_auto'
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl exposure_absolute=870
unknown control 'exposure_absolute'
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl -d /dev/video3 --set-ctrl exposure_auto_priority=1
unknown control 'exposure_auto_priority'
tommylkw@tommylkw-N3-Linux:~$

tlkw79, 2024-03-06:
This is what i found out.

tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl --list-ctrls --device /dev/video3

User Controls

                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=3 value=0
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=50
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=115 value=100
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1 (50 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=7
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=1

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=166 value=166 flags=inactive
                 focus_absolute 0x009a090a (int)    : min=0 max=1023 step=1 default=0 value=190 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=0 value=1

tlkw79, 2024-03-06:
One more thing, Can I know my UVC camera is running YUYY or MJPEG when connected in CC? 

I believe MJPEG far more better than YUYY.  Maybe you can provide an option to change this. 

Thanks.

tlkw79, 2024-03-15:
Any progress on this? Thanks a lot.

sajmons, 2024-03-15:
@tlkw79 No, no progres on this jet, sorry. I guess I'll have to get list of controly dynamically, so that user will only see those, that are supported by his camera. I just need to find time to implement this.

tlkw79, 2024-03-28:
this "Cameras are now detected dynamically and available in dropdown control" is good for windows user. But not for linux user on notebook with built in webcam. It will always start the built in webcam but not my UVC.

tlkw79, 2024-03-28:
Btw, in windows my UVC still failed to perform focus and other functions. I still depends on Sharpcap to act as video streamer.

sajmons, 2024-03-28:
> this "Cameras are now detected dynamically and available in dropdown control" is good for windows user. But not for linux user on notebook with built in webcam. It will always start the built in webcam but not my UVC.

You should change your camera in dropdown to see vich one will work. I dont know how to get default camera on Linux yet.

sajmons, 2024-03-28:
> Btw, in windows my UVC still failed to perform focus and other functions. I still depends on Sharpcap to act as video streamer.

Yeah, camera controls are work in progress. I don't have focusable camera at my hands, so I guess I will need your help here. Will contact you when I'm back to implementing controls.

sajmons, 2024-03-28:
This is your Linux laptop?
Can you run '''v4l2-ctl --list-devices'''?

tlkw79, 2024-03-29:
tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl --list-devices
Virtual Camera (platform:v4l2loopback-000):
        /dev/video0

HD User Facing: HD User Facing (usb-0000:00:14.0-5):
        /dev/video1
        /dev/video2
        /dev/media0

Depstech webcam MIC: Depstech w (usb-0000:00:14.0-8):
        /dev/video3
        /dev/video4
        /dev/media1

tlkw79, 2024-03-29:
> @tlkw79 Can you attach log too please?

Log for the UVC? I can't even connect to the UVC camera. I can only access to my HD User Facing webcam.

sajmons, 2024-03-29:
> > @tlkw79 Can you attach log too please?
> 
> Log for the UVC? I can't even connect to the UVC camera. I can only access to my HD User Facing webcam.

No, what I ment is log of CollimationCircles application. It's located in same folder as CC application in logs subfolder.

tlkw79, 2024-03-29:
This is the log file that you need.

sajmons, 2024-04-03:
> > @tlkw79 Can you attach log too please?
> 
> Log for the UVC? I can't even connect to the UVC camera. I can only access to my HD User Facing webcam.

@tlkw79 Thank you for log file.
You have UVC camera conected to that Linux laptop? V4L2 seams to detect only HD User Facing camera. How you conected UVC camera? With USB camle I supose? Can you use your UVC camera with your Linux machine?

tlkw79, 2024-04-03:
> @tlkw79 Thank you for log file. You have UVC camera conected to that Linux laptop? V4L2 seams to detect only HD User Facing camera. How you conected UVC camera? With USB camle I supose? Can you use [...]

UVC camera is connected to my laptop using USB cable. Before the latest version of CC, I can change the /dev/videoX. But in the latest version, I can't see the UVC camera option.

sajmons, 2024-04-04:
@tlkw79 I found possible bug in my code recognizing output from v4l2-ctl --list-devices command. But, when reviewing log that you send there is no 

```
Depstech webcam MIC: Depstech w (usb-0000:00:14.0-8):
/dev/video3
/dev/video4
/dev/media1
```

in the log. Are you sure, that your UVC camera is properly connected and recognized by v4l2-ctl?

tlkw79, 2024-04-04:
> tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl --list-devices Virtual Camera (platform:v4l2loopback-000): /dev/video0
> 
> HD User Facing: HD User Facing (usb-0000:00:14.0-5): /dev/video1 /dev/video2 /dev/media0
> 
> Depstech webcam MIC: Depstech w (usb-0000:00:14.0-8): /dev/video3 /dev/video4 /dev/media1

tommylkw@tommylkw-N3-Linux:~$ v4l2-ctl --list-devices
Virtual Camera (platform:v4l2loopback-000):
/dev/video0

HD User Facing: HD User Facing (usb-0000:00:14.0-5):
/dev/video1
/dev/video2
/dev/media0

Depstech webcam MIC: Depstech w (usb-0000:00:14.0-8):
/dev/video3
/dev/video4
/dev/media1

This is what I got when I plug in UVC camera into my laptop. The log is not capturing the UVC camera at all.

sajmons, 2024-04-18:
@tlkw79 there is new version available. Now your should be able to pick your UVC camera from list and control it's parameters.
Please let me know if it is working for you now?
https://github.com/sajmons/CollimationCircles/releases/tag/version-3.2.0-beta3

sajmons, 2024-05-09:
There is new version 3.2.0 beta5 available!

Changes are:
Finally camera controls should work for UVC cameras on Windows (through DShow) and Linux (through V4L2).
Performance improvements

Full change log: https://github.com/sajmons/CollimationCircles/compare/version-3.2.0-beta4...version-3.2.0-beta5

https://github.com/sajmons/CollimationCircles/releases

tlkw79, 2024-05-16:
Wow! latest version looks good in Linux and windows. But where is the Focus option? 

Thanks a lot.

sajmons, 2024-05-16:
@tlkw79 Can you be more specific about what is not working on Windows 11? I see there are errors in the logs, I'll correct that, but did camera stream showed or not?

> Wow! latest version looks good in Linux but not good in windows 11. Log files attached. [CC_Log.2024-05-16.txt](https://github.com/sajmons/CollimationCircles/files/15333756/CC_Log.2024-05-16.txt) [CC_Log.2024-05-16.1.txt](https://github.com/sajmons/CollimationCircles/files/15333757/CC_Log.2024-05-16.1.txt) [CC_Log.2024-05-16.0.txt](https://github.com/sajmons/CollimationCircles/files/15333758/CC_Log.2024-05-16.0.txt)
> 
> By the way, where is the Focus option?
> 
> Thanks a lot.

tlkw79, 2024-05-16:
@sajmons The camera doesn't stream at all. Just a small black screen pop up then the whole program hung there. Need to force it to close. 

Thanks.

sajmons, 2024-05-22:
> Wow! latest version looks good in Linux but not good in windows 11. Log files attached. [CC_Log.2024-05-16.txt](https://github.com/sajmons/CollimationCircles/files/15333756/CC_Log.2024-05-16.txt) [...]

Do you see focus slider in camera controls? Can you send me screenshot of camera controls dialog, from windows and linux machine? Thank you

tlkw79, 2024-05-22:
Hi,

I can see Focus option in Windows, but not in Linux.  But still, CC camera control not functioning in Windows. 

Thanks.
Tommy
[image]
[image]
[image]

sajmons, 2024-05-22:
Thank you Tommy. I found problem why Focus control is not visible on Linux. But I still don't know why it is not working? I'l have to investigate that further.

tlkw79, 2024-06-07:
Hi,

Any good news on software development? 

Thanks
Tommy

tlkw79, 2024-07-15:
Hi,

No new updates yet?

Best regards,
Tommy

sajmons, 2024-07-25:
> Hi,
> 
> No new updates yet?
> 
> Best regards, Tommy

Today I have released new version beta7. You can find it here: https://github.com/sajmons/CollimationCircles/releases

tlkw79, 2024-07-25:
> > Hi,
> > No new updates yet?
> > Best regards, Tommy
> 
> Today I have released new version beta7. More here: https://www.saimons-astronomy.com/1956214_new-release-of-collimation-circles-3-2-0-beta7

Hi,

Thanks for the latest update. 

1. Camera control works well in my kubuntu os but the resolution of the video display is bad. 
2. Camera control totally not working at all under windows 11. I also tried to disable my notebook internal webcam but still no hope on UVC camera control. The video resolution is good in Windows 11.

Best regards,
Tommy

sajmons, 2024-07-25:
Can you send some screenshots for linux and logs for Windows?

tlkw79, 2024-07-26:
> Can you send some screenshots for linux and logs for Windows?

This is some screenshots for linux. 

[image]

[image]

[image]

[image]

tlkw79, 2024-07-26:
> Can you send some screenshots for linux and logs for Windows?

Also some screenshots from windows 11 and logs.

[image]

[image]

[image]

[image]

[CC_Log.2024-07-26.txt](https://github.com/user-attachments/files/16386785/CC_Log.2024-07-26.txt)
[CC_Log.2024-07-26.5.txt](https://github.com/user-attachments/files/16386786/CC_Log.2024-07-26.5.txt)
[CC_Log.2024-07-26.4.txt](https://github.com/user-attachments/files/16386787/CC_Log.2024-07-26.4.txt)
[CC_Log.2024-07-26.3.txt](https://github.com/user-attachments/files/16386788/CC_Log.2024-07-26.3.txt)
[CC_Log.2024-07-26.2.txt](https://github.com/user-attachments/files/16386789/CC_Log.2024-07-26.2.txt)

tlkw79, 2024-07-26:
**_"Fix for executing v4l2 commands while on windows platform"_**

Just curious about windows implementation. Is it possible to use dshow instead?  

Tommy

sajmons, 2024-07-26:
> **_"Fix for executing v4l2 commands while on windows platform"_**
> 
> Just curious about windows implementation. Is it possible to use dshow instead?
> 
> Tommy

On windows dshow is used. It was a bug that I fixed not to execute v4l2 commands on windows any more. They had no efect, but probably some performance degradation. But as I said that's now fixed.

I'll see what can I do with controls not working on windows, for me they working normaly.

tlkw79, 2024-07-26:
> On windows dshow is used. It was a bug that I fixed not to execute v4l2 commands on windows any more. They had no efect, but probably some performance degradation. But as I said that's now fixed. [...]

That's weird. If your windows works, then how come mine can't? Is there any additional plugin or application needed that I need to install? 

for your info, in my laptop I got few others astro camera drivers as well. Will it be some sort of drivers compatibility happening here? 

Tommy

sajmons, 2024-07-26:
@tlkw79 I have succedded in repeting your case. I think that problem is in camera index. CC has curently hardcoded camera at index 0 thats equals to default camera on windows. Now I have two cameras and I'm able to reproduce error. I think that I will bw able to corect that issue soon. Till then stay tuned.

tlkw79, 2024-07-26:
Wohooo! Finally.....you did it! Will wait for your good news.  Well done.
 
Thanks a lot! 

Tommy

tlkw79, 2024-08-28:
Any news on the further release yet?

Thanks
Tommy

sajmons, 2024-10-16:
@tlkw79 try final stable release 3.2.0

tlkw79, 2024-10-16:
Good news! Later tonight I will test it out. Thanks a lot.

tlkw79, 2024-10-16:
Hi,

Now everything running fine now in Linux and Windows 11 as well. I can refresh the list and point to the UVC that I use now once plug in.  

Under Linux OS, the video still very lagging. But under windows OS, video display looks fine. 

Function list all working fine in both OS. 

I believe now you can implement the focus function back to the function list. This is more important to me :)

Thanks a lot. 

Regards,
Tommy

sajmons, 2024-10-16:
Tommy, that's really great news! Your help was priceless! I could not have done it without you.
Thank you very much for your effort.

I'm pretty sure that on Linux OS focus feature will be possible to implement (camera must support it), not so sure about Windows. I'll do my best!

What Linux distro are you using?

tlkw79, 2024-10-16:
> Tommy, that's really great news! Your help was priceless! I could not have done it without you. Thank you very much for your effort.
>
> I'm pretty sure that on Linux OS focus feature will be possible [...]

I am using Kubuntu OS.  Regarding to the video display lagging issue, will it be the resolution or framerate value?

Tommy

[13 more replies]

closed 2025-01-27 by sajmons