sajmons/CollimationCircles #23: Using Indi cameras

zsarosi, 2024-09-18:
Hi,
I am using CC on Linux (Ubuntu 22.04). Recently I have a ZWO ASI120mm mini camera, that as far as I know, has only Indi drivers under Linux. The camera beside it a great guiding camera, can be used with a lens as a collimation camera too. Unfortunately, I haven't find any way to map it to a v4l2 device. The only way to use it with CC is to display the camera stream in a Indi client e.g. EKOS, and overlay the CC.
Is there any plans to use Indi cameras directly in the software?
Thanks, and keep up the great job!
Zoltan

sajmons, 2024-09-19:
Hi Zoltan! Thank you for your suggestion. Yes It crossed my mind that it would be a great thing to support INDI cameras as well. Maybe in future versions.

sajmons, 2026-10-09:
@zsarosi I'm glad to announce that with version 5.0.0 you finally get your wish to come true. I have decided to switch to Alpaca for camera support. This way the CollimationCircles app gets support for all astronomical cameras that are supported by INDI or ASCOM platforms.

closed 2026-10-09 by sajmons

zsarosi, 2026-10-10:
Great news, many thanks!