sajmons/CollimationCircles #25: Does not find logitech usb webcam

Crowdedlight, 2025-01-09:
I wanted to test your amazing app out with a webcam and 3D printed adapter I have for my telescope. But the webcam does not show up in the dropdown menu in the latest release. Its a standard logitech 1080p webcam.

System: Windows 11
Webcam: Logitech 1080p
Version: 3.2.0 (Latest Release)
Log file: 
```
2025-01-09 00:47:12.8899: DEBUG MainWindowPosition changed to '789, 293'
2025-01-09 00:47:12.8899: DEBUG MainWindowPosition changed to '790, 293'
2025-01-09 00:47:12.8899: DEBUG MainWindowPosition changed to '792, 292'
2025-01-09 00:47:12.8899: DEBUG MainWindowPosition changed to '793, 292'
2025-01-09 00:47:16.3358: INFO Selected camera changed to ''
2025-01-09 00:47:16.4824: INFO Selected camera changed to ''
2025-01-09 00:47:16.6430: INFO Selected camera changed to ''
2025-01-09 00:47:16.7938: INFO Selected camera changed to ''
2025-01-09 00:47:29.9749: DEBUG MainWindowPosition changed to '-31994, -32000'
2025-01-09 00:47:34.7927: DEBUG MainWindowPosition changed to '793, 292'
2025-01-09 00:47:35.7742: INFO Saving application state to 'appstate.json'
```

Screenshot:
[image]

I saw the other issue that was closed about UVC camera focus control and downloaded and tried the beta software you linked there: https://github.com/sajmons/CollimationCircles/releases/tag/version-3.2.0-beta 

That kinda works as if I click "play" with ``dshow://`` as entry, then the webcam shows up, although performance starts to lag after a bit and it takes 10-20s from clicking play to the image shows up.
[image]
 
 Log from the beta version where the camera feed can be shown:
 ```
 2025-01-09 00:51:31.4369: DEBUG MainWindowPosition changed to '1343, 316'
2025-01-09 00:51:31.4574: DEBUG MainWindowPosition changed to '1344, 316'
2025-01-09 00:51:31.5337: DEBUG MainWindowPosition changed to '1344, 315'
2025-01-09 00:51:39.1537: DEBUG MainWindowPosition changed to '-31994, -32000'
2025-01-09 00:52:11.9236: DEBUG MainWindowPosition changed to '1344, 315'
2025-01-09 00:52:17.6639: DEBUG Shape 'Secondary' property 'IsVisible' changed to 'False'
2025-01-09 00:52:18.3717: DEBUG Shape 'Secondary' property 'IsVisible' changed to 'True'
2025-01-09 00:52:24.7586: INFO Saving application state to 'appstate.json'
```

My assumption is that the code that auto-detects the camera have some kind of bug not detecting my camera?

Crowdedlight, 2025-01-09:
I did some digging:

Changing the query line for getting entities to: 
```
SELECT * FROM Win32_PnPEntity WHERE (PNPClass = 'Image' OR PNPClass = 'Camera')
``` 
makes my webcam show up. However when then clicking play, it starts it, and `mediePlayer.play()` returns `true`, which should indicate it starts playing, but it only shows a black background and instantly stops itself again. I could not quite debug what happens. Was considering if it was a race condition where it does not wait on the lib to finish open the camera before checking itself and see that `isPlaying` is still false and stopping itself instantly again? 
When testing my webcam through VLC it also lags and takes a few seconds for it to start, but while using the webcam through applications like OBS or video conferencing it works fine with no lag. 

I do think I found why it might lag, at least in vlc. Apperently the Logitech webcams uses MJPG and not YUY2 which seems to be the default. I followed the approach on the last post on this page, and that fixed it in vlc. However not sure if that parameter can just be applied in the app and then it works, or there is something more at play here. 
https://forum.videolan.org/viewtopic.php?t=155105

sajmons, 2025-01-10:
@Crowdedlight information you provided here is incredible and it will help me a lot in making next release oc CC even better. Thank you very much. Indead there were some bugs in 3.2.0. 

I have applied yout patch SELECT * FROM Win32_PnPEntity WHERE (PNPClass = 'Image' OR PNPClass = 'Camera') to upcoming 4.0 version in Lincencing branch. https://github.com/sajmons/CollimationCircles/tree/Lincencing.

I have also removed command-line flag dshow-size wich caused your camera not working because unsupported resolution (I think).

Can you compile and run that version?

Crowdedlight, 2025-01-10:
Thanks for the quick response! 

I can confirm it works in terms of opening as shown on the video below:

https://github.com/user-attachments/assets/e8b0a016-4960-42f3-a927-b895bb9757e4

It still lags and have poor performance because it uses YUY2 instead of MJPG. However not sure if there is a way to auto-detect what the webcam wants to use in libVLC and set that. 
A workaround I found was adding ``:dshow-config=true` to the params in the code, as then when opening the webcam I get the dialog boxes from VLC where I can set the colour-space to MJPG and the camera performs much better as this video shows: 

https://github.com/user-attachments/assets/71a0a282-0c57-4e67-8d5b-7c8606034238

Not sure what would be a good solution here. Maybe add a checkbox next to camera dropdown where you can tick "advanced webcam options" that then allows the user to set those device settings themselves if their camera needs some specific handling?

sajmons, 2025-01-10:
Thank you for discovering that issue. And how to fix it. And yes, adding check box for advanced options would would probably be the best option for now to fix performance issues.

sajmons, 2025-01-10:
Can you send me your log file? I would like to see what is your camera DeviceID returned from ManagementObjectSearcher.

sajmons, 2025-01-10:
That's great news!

I'm not sure if MJPG instead of YUY2 is issue at all. My web cam uses YUY2 by default, but works the same with MJPG too.

For Service property please switch to latest 4.0 alpha1 version in Lincencing branch.
You can get it using this command:

```
git clone -b Lincencing https://github.com/sajmons/CollimationCircles.git
```

Then search for for line like below in log:
2025-01-10 21:47:04.3304: DEBUG DeviceName: 'Webcam C170, DeviceId: USB\VID_046D&PID_082B&MI_00\7&235F38C2&1&0000, Service: usbvideo'

Crowdedlight, 2025-01-10:
> That's great news!
> 
> I'm not sure if MJPG instead of YUY2 is issue at all. My web cam uses YUY2 by default, but works the same with MJPG too.

Fair. Just did not want to break other webcams if YUR2 was required for them ;-) 

> For Service property please switch to latest 4.0 alpha1 version in Lincencing branch. You can get it using this command:
>
> ```
> git clone -b Lincencing [...]

I checked branch to licensing and the log shows:
```
2025-01-11 00:17:18.1002: DEBUG DeviceName: 'HD Pro Webcam C920', DeviceId: 'USB\VID_046D&PID_0892&MI_00\7&33F489B&2&0000', Service: 'usbvideo'
```

sajmons, 2026-10-09:
With version 5.0.0, I have decided to switch to Alpaca for camera support. This way the CollimationCircles app gets support for all astronomical cameras that are supported by INDI or ASCOM platforms.

@Crowdedlight UVC cameras are not supported anymore. Collimation circles now support all astronomical cameras that are supported by the INDI or ASCOM platform.

closed 2026-10-09 by sajmons